projecthorus/radiosonde_auto_rx #1008: Capturing UDP broadcasts within the same Docker network where the radiosonde resides.

tushroy, 2025-05-24:
I found it difficult to capture the UDP broadcast within the same Docker network, except by using the host network — which may not always be feasible.
What I want to achieve is to extract the raw data from `radiosonde_auto_rx` and either store it in a different database or process it with a separate system.
Since I'm having difficulty achieving this, are there any plans to provide an alternative method for accessing the data?
For example, could `radiosonde_auto_rx` support HTTP POST of the raw JSON data to a user-defined endpoint configured in `station.config`, allowing other systems to use the data?

tushroy, 2025-05-24:
> At the moment, that's not planned. I do appreciate that the UDP data is a bit difficult to get access to, and maybe I should look at setting up something like MQTT alongside. To get access to the data _right now_, you could possibly make use of the web-sockets (sockert.io) interface that the web interface uses.

Thank you Mark, for considering it.
Even MQTT will do. I will have a look into the web-sockets meanwhile.

darksidelemm, 2025-05-24:
I suspect that using the python socket.io client might allow accessing the messages sent from the server: https://python-socketio.readthedocs.io/en/latest/client.html

I think pretty much the same data that's sent out via UDP is sent to the web interface live via socket.io

JhPj4VOb2h, 2026-11-10:
> Thank you Mark, for considering it. Even MQTT will do. I will have a look into the web-sockets meanwhile.

A bit late to the party, but I would also love an MQTT interface. Mostly to notify my Home Assistant system to send me a message when a sonde has been received nearby.

darksidelemm, 2026-11-10:
So I'm fairly unfamiliar with MQTT for this kind of application.

Would auto_rx have an internal MQTT server, or would it connect out to another MQTT server? 
I'm guessing you would want the same kind of data that is emitted via the UDP broadcast.

JhPj4VOb2h, 2026-11-10:
auto_rx would be a client connecting to another MQTT server. That server feeds Home Assistant which would handle the notifications.

As a more general notification system, Apprise (https://github.com/caronc/apprise), could be a way to let people choose their own notifications from auto_rx. Might help with #8 and #624. Worth adding as a feature request?

darksidelemm, 2026-11-10:
Ive been a bit loathe to put a more generic notification system like this in here. If I did, it would be used as a replacement for the email notification functionality, for notification of new observed launches or landings. It wouldn't be an alternative to sending out *all* the received telemetry.
I need to understand a bit more how something like apprise would be configured in our station.cfg file, and how generic such a configuration can be (e.g. do some apprise integrations need more parameters?)

Anyway, connecting to a MQTT server does sound very doable. What would be the configuration settings required be?
- Server hostname
- Server port
- Presumably some kind of topic setting? Push? 
Do you have any examples of how this kind of thing is configured in other software?

dbrooke, 2026-11-10:
You'll probably also want user and password in config

I added basic MQTT support to my UKHASnet Pi gateway in the below commit, it's in C but may be of some help

https://github.com/dbrooke/UKHASNetPiGateway/commit/6fd7e76c397a4faaab261228159316f7f0fd8ba1

xssfox, 2026-11-10:
You could follow sondehub approach in sending sondehub standard format to a topic per serial number

darksidelemm, 2026-11-11:
Yeah, that'd be a good idea. I'll need to reshuffle a few things to be able to get the sondehub data reformatter accessible from other modules, but it should be doable.

JhPj4VOb2h, 2026-11-11:
> Ive been a bit loathe to put a more generic notification system like this in here. If I did, it would be used as a replacement for the email notification functionality, for notification of new [...]

Server connection would need a hostname/ip, port (defaults to 1883), username, and password. TLS setup would be a later step and would depend on the user's MQTT broker settings.
Main topic setting would be something like 'radiosonde_auto_rx' with values in messages under that.
radiosonde_auto_rx/_sond_id_/_measurement_type_/_value_
It would be up to the whatever subscribed to the that topic on how to display.

https://pypi.org/project/paho-mqtt/ for a client library. https://www.emqx.com/en/blog/how-to-use-mqtt-in-python is an OK tutorial about using that library. 

Birdnet-Pi (https://github.com/Nachtzuster/BirdNET-Pi) for the more generic notification system. I just haven't figured out where in the web code all of the setup is yet, but you don't need all of the notifications available.

darksidelemm, 2026-11-11:
OK, I don't think i'd be going to the extent of publishing different measurement types as different topics. It'd be one topic per radiosonde I think....