projecthorus/radiosonde_auto_rx #1014: Radiosonde Support - Changfeng CF-06 series

DJRaov, 2025-07-24:
Hi again! I just got around to receiving radiosondes again, and I heard that Kazakhstan (or Kostanay, at the very least) has started launching CF-06 sondes alongside DFMs. Just now I got a decent recording, how hard would it be to add support? It would be nice to see them get supported in auto_rx eventually.

Sample link: https://ub8qbd.satdump.org/files/cf06-usb-07242025.wav

DJRaov, 2025-07-24:
It was indeed recorded in USB, mostly because SNR was too low in the beginning and I felt the decoding would be better if it wasn't demodulated already. I will record a proper 48kHz IQ tomorrow.

hb9evw, 2025-07-24:
OK, thanks for the reply. I understand. It would be great if you can record one in 48KHz IQ! Best regards and 73

rs1729, 2025-07-25:
If it is the same CF-06 as here
https://github.com/rs1729/RS/tree/test/uaii2022
I prepared a decoder with JSON output back then
https://github.com/rs1729/RS/tree/test/cn
Probably some adjustments for auto_rx could be made and detection in dft_detect added.

But let's see a recent IQ file.

DJRaov, 2025-07-25:
> IQ recordings from a radiosonde on the ground at a known location.
Sadly I fear that is going to be impossible to obtain without buying one directly, the launch station is about 350km away from me.
Maybe someone from other countries where they are being launched can contribute a sample.

Today's sample: https://ub8qbd.satdump.org/files/cf06-raw-07252025.wav

hb9evw, 2025-07-25:
Thanks for the sample. No worries, 350km is very far for good SNR. 
Anyway, many thanks for your efforts! Best 73s

DJRaov, 2025-07-26:
Got another recording, much better SNR this time: https://ub8qbd.satdump.org/files/cf06-raw-07262025.wav

rs1729, 2025-07-26:
This is not the CF-06AH that was used in UAII 2022.
This is a "slower" version with less data that can be decoded by the other decoder in
https://github.com/rs1729/RS/tree/test/cn
`unk600bd_20240130.c`
Perhaps a pilotsonde?
```
 13:00:22.0   lon: 63.82644  lat: 53.42400  alt: 26977.9   vH:4.98  D:286.8  vV:7.06   [6329]   [1011:1011]  [OK]
 13:00:23.0   lon: 63.82637  lat: 53.42402  alt: 26985.1   vH:4.91  D:312.5  vV:5.51   [6330]   [4B2F:4B2F]  [OK]
 13:00:28.0   lon: 63.82613  lat: 53.42403  alt: 27013.2   vH:6.70  D:224.8  vV:4.11   [6335]   [F316:F316]  [OK]
 13:00:30.0   lon: 63.82597  lat: 53.42395  alt: 27023.2   vH:7.39  D:255.4  vV:6.66   [6337]   [4189:4189]  [OK]
 13:00:31.0   lon: 63.82586  lat: 53.42394  alt: 27029.9   vH:7.12  D:268.6  vV:5.96   [6338]   [141F:141F]  [OK]
 13:00:34.0   lon: 63.82560  lat: 53.42397  alt: 27046.9   vH:2.64  D:265.7  vV:6.65   [6341]   [E561:E561]  [OK]
 13:00:39.0   lon: 63.82518  lat: 53.42384  alt: 27076.6   vH:13.05  D:263.9  vV:5.92   [6346]   [3D08:3D08]  [OK]
 13:00:40.0   lon: 63.82497  lat: 53.42383  alt: 27082.5   vH:13.63  D:271.8  vV:4.26   [6347]   [6946:6946]  [OK]
 13:00:41.0   lon: 63.82478  lat: 53.42384  alt: 27087.1   vH:11.87  D:279.7  vV:6.15   [6348]   [C847:C847]  [OK]
```
(note: lon/lat/alt, i.e. N53.42 E63.82 27km)
Does this look reasonable?

DJRaov, 2025-07-26:
Yep, coords look reasonable. I noticed it sounds slower too, and [this spec sheet](https://www.sstc.org.in/images/data-sheet/gps-radiosonde-sounding-system.pdf) supports the slower speed. Configurable perhaps? 
[This Alibaba listing](https://www.alibaba.com/product-detail/CF-06-GPS-Radiosonde-Balloon-Radiosonde_60296392592.html) suggests that CF-13 is also part of the series. Maybe I am hearing a CF-13 and not a CF-06, after all?

rs1729, 2025-07-26:
Don't know this 600 baud variant. It is a mix of CF-06 and HT-03 with less data, and 4 times slower, but similar bandwidth (i.e. high modulation index). I don't know how these radiosondes are connected, but the frame format looks similar as if they are related.

Now I know three recordings. The last 3 bytes are constant in these recordings, perhaps the serial number? A serial number would be important for auto_rx. Would be nice to see an image that confirms a serial number.

[cf06-raw-07262025_log_OK2.txt.gz](https://github.com/user-attachments/files/21449657/cf06-raw-07262025_log_OK2.txt.gz)

darksidelemm, 2025-07-27:
For inclusion into auto_rx, I definitely want to know if there's a serial number transmitted in the telemetry, and for that to be used. 
If not, we'd have to go down the road of the imet generated serial number approach, but that's really a last resort.

rs1729, 2025-07-28:
The last recordings last three bytes: 03 45 3A
Don't think it's a date, but the numbers are progressing a bit like hours, however constant for each recording. Too bad you could not record the 00Z rollover.
```
 00:27:52.0   lon: 63.59172  lat: 53.44337  alt: 15280.3   vH:6.88  D:28.4  vV:4.01   [4163]     [3A89:3A89]  [OK]
```
At 00Z altitude was around 9km I guess.

If it is a serial number, it is still unclear, how it translates to the actual number written on the radiosonde. CF-06AH and HT-03G have there 4 byte serial numbers at the beginning of the frame, but different formats.

Maybe this one (600 baud) is an earlier version of the CF-06? Or maybe it has a different model name. Would be good to know, so a type name can be given.

darksidelemm, 2026-09-19:
Fixed in 1.9.0

closed 2026-09-19 by darksidelemm