projecthorus/radiosonde_auto_rx #1016: rtlblog v4 stick  problems on scanning

whallmann, 2025-07-28:
Hey
are there some known Problems with the new rtlblog v4 sticks?
in autorx there is a problem after some scanns.
Other apps like dxlaprs c# e.g. dont tell problems 

here are some log messages:

Scanner (RTLSDR 00000002) - rtl_power reported error: Found 3 device(s): 0: Realtek, RTL2838UHIDIR, SN: 00001003 1: RTLSDRBlog, Blog V4, SN: 00000002 2: Nooelec, NESDR SMArt v5, SN: 56431378 Using device 2: Generic RTL2832U OEM Number of frequency hops: 3 Dongle bandwidth: 2177777Hz Downsampling by: 1x Cropping by: 25.00% Total FFT bins: 12288 Logged FFT bins: 9216 FFT bin size: 531.68Hz Buffer size: 16384 bytes (3.76ms) Reporting every 20 seconds usb_claim_interface error -6 Failed to open rtlsdr device #2.

run in scanmode for some time then stops.
The other two older sticks Work well.  The drivers used from the rtlblog site so they should play with V4 type

whallmann, 2025-07-28:
remark:  the retrys for reset on usb 001/003 dont get succes. After restart the autorx it works again fir short time

darksidelemm, 2025-07-28:
USB claim error indicates that something else is using the RTLSDR. Check
carefully that there isn't another instance of autorx running, or some
other software that is using the RTLSDR.

whallmann, 2025-07-28:
I am pretty sure that there is no other instance running. It only affects
the blogV4 stick not the others.
Hmmm.

whallmann, 2025-07-28:
> Looking at your log closely, it looks like you have more than one RTLSDR
> device. Adding line breaks for clarity:
>
> Scanner (RTLSDR 00000002) - rtl_power reported error:
> Found 3 device(s):
> 0: Realtek, RTL2838UHIDIR, SN: 00001003
> 1: RTLSDRBlog, Blog V4, SN: 00000002
> 2: Nooelec, NESDR SMArt v5, SN: 56431378
> Using device 2: Generic RTL2832U OEM
>
> It looks like you have specified device ID '2' in your config file, but
> it's using the device with index 2 instead (not the RTLSDR v4). With this
> many RTLSDRs, things can get weird with how rtl_fm/rtl_power interpret the
> device id field.
> I would suggest changing the serial number to something non-numeric, or a
> much higher numbers like you have done with the other two.

Yes Mark you got it, but this is ok.  I droped 3 Sticks to autorx.  the
parameter -d counts from 0 up. But the #2 means it is the second Stick (
-d1). Other messages in the stack says its serial number 00000002. After
losing contact this stick is not longer listed in the webbrowser. So all
hints are pointing to the blogV4 Stick.

Meanwhile i extracted it from the sdr list in config and run it on a
parallel running  dxlAprs Chain as single stick. There it works without
Problemes again.

Only the rtlpower means to have some missbehaivior.  rtlpower is used as
scanner only? So this stick has assigned the lowest serial and always
assigned as scanner first. maybe i can decide to not use it as scanning
device? I think not.

Pretty mess is only that this blogV4 is the sensitivest stick of all of
them.

I dont have more Ideas to test at this moment.

cu

darksidelemm, 2025-07-28:
auto_rx doesn't use just one SDR for scanning. It uses them as a 'pool', assuming they are all the same, and all connected to the same antenna. 

Please try changing the serial number to something else. With 3 SDRs, a serial number of '2' will be interpreted as the third device (indexed from 0) which is clearly causing problems.

whallmann, 2025-07-29:
Ok after knew this details i will rename its serial.  I tried this before
but I need physical access to it. unfortunately it is on a radio tower.
Software resets of USB connectors and reboots didnt helped to get the new
serial visible. It seems someone has to unplug it personally.  So i have to
wait.

thanks
Wolf